ARM-software/bsa-acs #282: ./linux_bsa_acs.sh failed

QTAQ, 2024-04-16:
Hi!
I am building bsa-acs following the README says:

### 1. build linux kernel with bsa-acs patch.

- git clone https://git.gitlab.arm.com/linux-arm/linux-acs.git linux-acs

- git clone https://github.com/ARM-software/bsa-acs.git bsa-acs

- git clone https://github.com/torvalds/linux.git -b v6.7

- export CROSS_COMPILE=<GCC13.2 toolchain path> pointing to /bin/aarch64-linux-gnu-

- git apply <local_dir>/linux-acs/kernel/src/0001-BSA-ACS-Linux-6.7.patch to your kernel source tree.

- make ARCH=arm64 defconfig && make -j $(nproc) ARCH=arm64

when doing this, compile is failed. reason is that drivers/iommu/bsa-dma-iommu.c has include dma-iommu.h,
but the code is
  **include <dma-iommu.h>**
Is that bsa-dma-iommu.c wants to include dma-iommu.h in /drivers/iommu ? So this should be 
  **include "dma-iommu.h"**
After modify this include, compile will success.

### 2. build bsa-acs.ko

- cd <local_dir>/linux-acs/files

- export CROSS_COMPILE=/bin/aarch64-linux-gnu-

- export KERNEL_SRC=<linux_kernel_path>

- ./bsa_setup.sh <local_dir/bsa-acs>

- ./linux_bsa_acs.sh

compile is success, but modpost is failed:
bsa_acs_pal.o:
ERROR: modpost: missing MODULE_LICENSE() in /root/linux-acs/acs-drv/files/platform/pal_linux/files/bsa_acs_pal.o
ERROR: modpost: "g_print_level" [/root/linux-acs/acs-drv/files/platform/pal_linux/files/bsa_acs_pal.ko] undefined!

bsa_acs_val.o:
ERROR: modpost: missing MODULE_LICENSE() in /root/linux-acs/acs-drv/files/val/bsa_acs_val.o
ERROR: modpost: "pal_mem_virt_to_phys" [/root/linux-acs/acs-drv/files/val/bsa_acs_val.ko] undefined!
ERROR: modpost: "g_num_tests" [/root/linux-acs/acs-drv/files/val/bsa_acs_val.ko] undefined!
WARNING: modpost:
suppressed 88 unresolved symbol warnings because there were too many)

This seems to be an issue with the build order? Or perhaps a problem with modpost?

chetan-rathore, 2024-05-03:
Hi @QTAQ,

We have updated the header file changes in the kernel patch based on 6.8 version, which is latest supported at our side.
https://gitlab.arm.com/linux-arm/linux-acs/-/blame/master/kernel/src/0001-BSA-ACS-Linux-6.8.patch?ref_type=heads#L103

The modpost related error/warning can be ignored, BSA or SBSA kernel module compilation have intermediate steps of compiling pal, val and test layers. The undefined warning are for functions which are defined in other layer. The final compilation steps of kernel module generation should not have any undefined.

Please try to build with latest kernel v6.8, READMe updated with the required steps.

Thanks,
ACS team

chetan-rathore, 2024-05-07:
Hi @QTAQ, 

Please reopen the issue, if any build issues faced with linux 6.8 bsa linux acs compilation.

Thanks,
ACS team

closed 2024-05-07 by chetan-rathore