ARM-software/bsa-acs #303: Add few more prints to PE test 1 (Check Arch symmetry across PE)

sunnywang-arm, 2024-05-23:
We want to know the inconsistent value of PE registers for debugging. However, now the message only shows the masked out value. 

For verbosity message, add few prints to show values of registers right after the line https://github.com/ARM-software/bsa-acs/blob/main/test_pool/pe/operating_system/test_os_c001.c#L172.

For default mesasge, add prints right before the line https://github.com/ARM-software/bsa-acs/blob/main/test_pool/pe/operating_system/test_os_c001.c#L174.

sunnywang-arm, 2024-05-29:
Chetan found an issue with line https://github.com/ARM-software/bsa-acs/blob/main/test_pool/pe/operating_system/test_os_c001.c#L248 while checking this improvement. The correct value is held by cache_list and not by rd_data_array so the message "Current PE value = xxxxxxxxxx" could be misleading. 

For adding prints, Test 1 is multi-PE test, but On EDK2/UEFI you can’t use any protocols to print message on secondary PEs as the code is not MP safe. Therefore, ACS team will look into the solution that make the memory buffer bigger and save values from all PEs then print them all in the primary PE.

chetan-rathore, 2024-06-19:
Hi @sunnywang-arm,

ACS team is working on the changes and will upstream the changes soon.

Thanks,
ACS team

PR #322 (2024-07-04): Secondary PE Print for pe test 01
The PE test - 01 has been updated to check for all fails in registers and cache.
The secondary PE registers value can be stored in a structure and can be printed by Primary Pe for both fail and info cases.

Fixes #303, #304, #310
closed 2024-07-04 by chetan-rathore