ARM-software/bsa-acs #305: Stop listing iEP_EP/RP entries in PCIE_INFO

hrw, 2024-05-24:
On my test setup I never have iEP_EP/RP entries listed so I decided to check what I need to have for it.

```
 PCIE_INFO: Number of BDFs found      :    10
 PCIE_INFO: Number of RCiEP           :    2
 PCIE_INFO: Number of RCEC            :    0
 PCIE_INFO: Number of EP              :    1
 PCIE_INFO: Number of RP              :    4
 PCIE_INFO: Number of iEP_EP          :    0
 PCIE_INFO: Number of iEP_RP          :    0
 PCIE_INFO: Number of UP of switch    :    1
 PCIE_INFO: Number of DP of switch    :    1
 PCIE_INFO: Number of PCI/PCIe Bridge :    0
 PCIE_INFO: Number of PCIe/PCI Bridge :    1
```

Turned out that BSA ACS (SBSA ACS uses same code) does not handle them at all. In [`var/common/src/acs_pcie.c` in line 892](https://github.com/ARM-software/bsa-acs/blob/23433ca89680fa76a3c4d2df5a33fed515bc557c/val/common/src/acs_pcie.c#L902) there is `if (val_pcie_is_onchip_peripheral(bdf))` check which leads to [`pal/uefi_acpi/common/src/pal_pcie.c` line 492](https://github.com/ARM-software/bsa-acs/blob/23433ca89680fa76a3c4d2df5a33fed515bc557c/pal/uefi_acpi/common/src/pal_pcie.c#L492) where `0` is return.

So there are no iEP_EP/RP entries on all UEFI ACPI systems.

Can we get rid of listing them then?

kiranronad, 2024-06-14:
Thanks for observation.
 
Generally, ACS listing all different type of PCI/PCIe devices in the system independent of BSA or SBSA.
By default, ACS prints '0' for number of iEP_EP and iEP_RP entries for both of BSA and SBSA as these device types(port type) can not be identified with configuration register space.
 
Partner need to update the PAL interface if system has iEP_EP and iEP_RP devices(BDFs)  which intern reflects in number of respective devices in printouts.
PAL Interface link:
https://github.com/ARM-software/bsa-acs/blob/23433ca89680fa76a3c4d2df5a33fed515bc557c/pal/uefi_acpi/common/src/pal_pcie.c#L492
 
So, we believe no ACS change required.
 
Kindly let us know if more information needed here.

hrw, 2024-06-14:
ok

closed 2024-06-14 by hrw