ARM-software/bsa-acs #308: PE test 1 doesn't check last few PE registers in reg_list[]

sunnywang-arm, 2024-05-29:
The value of NUM_OF_REGISTERS should be updated to match the number of registers in reg_list []. There are more than 32 registers in reg_list[]. 
https://github.com/ARM-software/bsa-acs/blob/8d2382ad2be86ee2598dc69703e11c9315c3cf2b/test_pool/pe/operating_system/test_os_c001.c#L25

magicse7en, 2024-06-17:
some register reads are commented out, which should also be updated.
https://github.com/ARM-software/bsa-acs/blob/main/val/common/src/AArch64/PeRegSysSupport.S#L394C9-L394C23

sunnywang-arm, 2024-06-20:
Thanks for the feedback, @hrw and @magicse7en. Agree with your comments. I created another task to update the list. 
https://github.com/ARM-software/bsa-acs/issues/318

chetan-rathore, 2024-10-03:
PR #342 has enabled further 5 registers which were mentioned in test case, but were either commented out or not executing due to lesser value of NUM_OF_REGITER.

sunnywang-arm, 2024-10-03:
Thanks, @chetan-rathore. Closing the issue as it got fixed by PR https://github.com/ARM-software/bsa-acs/pull/342.

closed 2024-10-03 by sunnywang-arm

hrw, 2024-10-17:
Please add ID_AA64MMFR3_EL1 register. We got cpu which has Spec_FPACC bits set.