ARM-software/bsa-acs #318: Many PE registers haven't been checked by PE test 1

sunnywang-arm, 2024-06-20:
There is a good catch from https://github.com/ARM-software/bsa-acs/issues/308 

All (more than 100) of the PE registers, except the ones in the exception list in the BSA Appendix A should be checked by PE test 1, but now we just check 40 PE registers. We need to complete the list. https://github.com/ARM-software/bsa-acs/blob/8d2382ad2be86ee2598dc69703e11c9315c3cf2b/test_pool/pe/operating_system/test_os_c001.c#L52

sunnywang-arm, 2024-12-06:
Hi @chetan-rathore  Any update for this issue? Do we have an estimated time of fix?

rajatgoyal47, 2024-12-12:
Hi Sunny,

We have started working on this and will be adding the registers incrementally. Currently, we are adding a few ID_AA64* related registers and will upstream the changes after verification.

Regards,
Rajat
ACS Team

sunnywang-arm, 2024-12-12:
Hi @hrw
Besides the registers you mentioned on https://github.com/ARM-software/bsa-acs/issues/308#issuecomment-2419010691, any other registers you think we need to add first.