ARM-software/bsa-acs #323: BSA test "601 : USB CTRL Interface" need to be revisited

sunnywang-arm, 2024-07-05:
This has been discussed offline. Create an issue here for bsa-acs users' information. 

The corresponding rules below in BSA spec are conditional requirements. 
- B_PER_01 : If the system has a USB2.0 host controller peripheral it must conform to EHCI v1.1 or later.
- B_PER_02 : If the system has a USB3.0 host controller peripheral it must conform to XHCI v1.0 or later.

Now the BSA test 601 only allows the system to have EHCI and XHCI USB controllers. A system having a USB 1.x controller will cause the test 601 to fail, which is wrong. We need to fix the issue.  For DT based system, we already have a pull request for this.   https://github.com/ARM-software/bsa-acs/pull/317. For ACPI based system, we still need make a change. 

Also, the test doesn't completely check the compliance of B_PER_01 and B_PER_02. It only checks the type of USB controller, so we'll need to make some changes to check the USB 2.0 and 3.0 controller's EHCI or XHCI version.

chetan-rathore, 2024-07-08:
Hi @sunnywang-arm,

The test will be updated to 
* fail only if the system doesn't have any EHCI 1.1 or XHCI controller.
* EHCI version check of 1.1 will be added.

Thanks,
ACS team

sunnywang-arm, 2024-07-09:
@chetan-rathore  

As we offline discussed, B_PER_01 and B_PER_02 are conditional requirements, so If the system doesn't have any EHCI or XHCI controller, the result should be SKIPPED rather than FAIL.

Also, the purpose of BSA test 601 is to catch the case where non-standard USB 2.0 and 3.0 controllers (not compliant with EHCI and XHCI specs) are being used for OS boot and installation.

Therefore, the test will be updated to:   
- SKIPPED if the system doesn't have any EHCI or XHCI controller. 
- FAILED only if the present EHCI and XHCI controllers are not compliant with EHCI and XHCI specs. 
    - Check all USB devices' PCIe interface type to find all EHCI and XHCI controllers in the system. 
    - Read and check all found EHCI and XHCI controllers' registers that are defined in EHCI or XHCI specs. If the register value doesn't match the value defined in spec, fail the test.