ARM-software/bsa-acs #327: Does system ready code support gic v4.1

LaughingLeeLife, 2024-07-24:


rajatgoyal47, 2024-07-31:
Hi,

Currently GIC Related code in ACS is written as per GICv3 which should be backward compatible with GICv4.1, but it does not have support for extra features added in GICv4.1.
Could you please let us know if you've encountered any unexpected behaviour/failures on a v4.1 system? If so, please provide details.

Regards,
ACS Team

rajatgoyal47, 2024-08-08:
Hello,

Details about GICD_BASE address and GICC_BASE address are not defined by GIC Architecture; however the register offsets from these base addresses are specified by the GIC architecture. 

Information regarding different base addresses should be available in the system memory map and will vary depending on the implementation.

Thanks,
ACS Team

chetan-rathore, 2024-08-26:
Hi @LaughingLeeLife,

Please let us know if you have any further queries or can the ticket be closed.

Thanks,
ACS team

closed 2024-09-03 by chetan-rathore