ARM-software/bsa-acs #337: BSA tests 605 and 862 rely on val_dma_create_info_table but Val code never calls it

camwilde77, 2024-09-18:
Both BSA tests 605 and 862 call val_get_dma_info. 

The header of that function states:
  @brief  Single entry point to return all DMA Controller related information.
          1. Caller       - Test Suite
          **2. Prerequisite - val_dma_create_info_table**
  @param  info_type - Type of DMA controller whose index needs to be returned
  @param  instance  - id (0' based) for which the info has to be returned

Grepping through the BSA repo, the val_dma_create_info_table is never called and thus no DMA controllers are found.

Shouldn't the test setup or the val side of things call this function?

rajatgoyal47, 2024-09-26:
Hello,

Both tests are executed in a Linux environment or on bare metal.

For more details about the Linux environment, you can refer to the following link: https://gitlab.arm.com/linux-arm/linux-acs/-/blob/master/acs-drv/files/bsa_acs_drv.c#L107

These tests are skipped in a UEFI environment, which is why the function is not being called.

Regards,
ACS Team

camwilde77, 2024-09-30:
We are doing bare metal and we still don't see a call in the test suite. Where do we expect a call?

rajatgoyal47, 2024-10-01:
Hi,

Thanks for reporting this issue.
We will work on resolving it and keep you updated.

Regards,
ACS Team

rajatgoyal47, 2024-10-10:
Hello,

We will be upstreaming the change to call val_dma_create_info_table.

To help you better, we need some information regarding the DMA controllers in your environment. 
The two tests require details about DMA controllers, memory attributes, as well as the relevant DMA drivers.

Could you please provide details about the DMA controllers on your platform? Additionally, it would be helpful to know if the necessary DMA drivers are already in place.

Best Regards,
ACS Team

PR #375 (2024-10-30): Fixes #337:  val_dma_create_info_table in BM
Fixes #337 
- Call to DMA create info table is needed in BM to populate the DMA table
- The details to be populated in the platform config files
commit 62cbacd7 (2024-10-30): Fixes #337:  val_dma_create_info_table in BM (#375)

- Call to DMA create info table is needed in BM to
  populate the DMA table
- The details to be populated in the platform config
  files
closed 2024-10-30 by chetan-rathore