ARM-software/bsa-acs #338: GIC tests seems to rely on MSI-X feature being enabled for v3 + ITS

camwilde77, 2024-09-18:
From my understanding, the GIC (GIV v3 + ITS) tests can be set up as MSI or MSI-X. v3 + ITS can have full MSI-X support at the RP, but doesn't have to use it. You could also have an endpoint that supports it (or not). As far as I can tell, the BSA code expects that MSI-X is always supported for GIC v3 + ITS, but a system could still decide to not do that. Shouldn't these tests default to MSI when the RP compatibility register shows MSI-X as not enabled? This seems to affect tests 904, 911. 912, and 913.

sunnywang-arm, 2024-09-30:
I agree with the comment here. BSA PCIe exerciser tests should check both MSI and MSI-X capable endpoint devices rather than only MSI-X capable endpoint devices. 

@gowthamsiddarthd and @chetan-rathore  Can we get ACS team to fix the issue?

rajatgoyal47, 2024-10-10:
Hi,

Thank you for reporting this. 
We are working on fixing and adding support for MSI in the tests. 

Regards,
ACS Team

cdwilde, 2025-01-24:
Do we have a timeframe for this fix?

Sujana-M, 2025-03-05:
Hi @cdwilde @camwilde77 ,

The changes for MSI support are at commit 
BSA:https://github.com/Sujana-M/bsa-acs/commit/004eca002ffe397450e2f175fef1602216ff98cd 
SBSA: https://github.com/ARM-software/sbsa-acs/commit/07c1aaa33f204c2e7d7f4b8ff919d0ff0ab80a1e.

The changes can be cherry-picked at your end. Please let us know if this provides the coverage.

Thanks,
ACS Team

cdwilde, 2025-04-09:
I have been using this in my repo. When will it be officially released?

Sujana-M, 2025-04-11:
Hi @cdwilde,

Please confirm if the above patch has been verified at your end and provides the coverage. We will merge it once it is confirmed.

Thanks,
Sujana

cdwilde, 2025-04-14:
@Sujana-M, I don't have a good way to test the exerciser tests, but the non-exerciser tests look good. Please go ahead and merge.

closed 2025-07-01 by prashymh