ARM-software/bsa-acs #355: Baremetal FVP RDN2 platform data issue - Invalid MPAM data failed test 16

sunnywang-arm, 2024-10-14:
The invalid MAPM data (platform_mpam_cfg) in pal\baremetal\target\RDN2\sbsa\src\platform_cfg_fvp.c and the value of PLATFORM_MPAM_MSC_COUNT in pal\baremetal\target\RDN2\sbsa\include\platform_override_sbsa_fvp.h failed the test 16.  

16 : Check MPAM LLC Requirements           
       START
       MSC count = 0
       MSC count is 0
       Failed on PE -    0
       S_MPAM_PE
       Checkpoint --  1                           : Result:  FAIL
       END

chetan-rathore, 2024-10-24:
The RDN2 ACPI tables or TRM doesn't capture details of MPAM MSC count and address bases, hence the mpam related information is not filled in baremetal pal files.

chetan-rathore, 2025-02-18:
Closing as this is not ACS bug, once the RDN2 model adds required tables it will be verified internally

closed 2025-02-18 by chetan-rathore