ARM-software/bsa-acs #358: Fix formatting in test 102

hrw, 2024-10-17:
```
 102 : Mem Access Response in finite time
       START
       Device Mem Address : 0x10000000
       Device Mem Address : 0x60010000Instance 0x2 not found for memory type 0x1000

       Error in getting dev mem for index 2
       Normal Mem Address : 0x10000000000
       Normal Mem Address : 0x101F86D0000
       Normal Mem Address : 0x101F8730000
       B_MEM_01
                                       : Result:  PASS
       END
```

chetan-rathore, 2024-10-17:
Hi @hrw,
Your findings are making the suite better. 
Thanks for running BSA/SBSA at your end.

We welcome PR also, looking forward to it.

hrw, 2024-10-17:
(S)BSA ACS is developed outside of this repo in a way which I find uncomfortable. Large code drops instead of granular commits, random whitespace changes between changes etc.

That's why I decided over year ago to report issues rather than sending pull requests.

https://github.com/ARM-software/sbsa-acs/issues/326 has more comments on it.

closed 2024-10-24 by chetan-rathore