ARM-software/bsa-acs #359: Add a way to run test 835 on system with only network/storage cards

hrw, 2024-10-17:
Test 835 skips all network and storage devices:

```
 835 : Check Function level reset
       START
       Skipping for BDF - 0x100  Classcode is : 0x2
       Skipping for BDF - 0x10000  Classcode is : 0x2
       Skipping for BDF - 0x80000  Classcode is : 0x2
       Skipping for BDF - 0x90000  Classcode is : 0x1
       No EP type device found with PCIe Express Cap support. Skipping test
       PCI_SM_02
       Checkpoint --  1                           : Result:  SKIPPED
       END
```

Then says that there were no PCI Express End Point devices.

Comment in code says:
```c
      /* Skip check for Storage devices as the
       * logs will not be stored if FLR is done
       * Skip for ethernet controller as device
       * init can get corrupted when FLR is done */
```

What kind of "device init" can get corrupted when BSA ACS tells at the end "Reset the system"?

Maybe there could be some way added to force run on test 835 (other than changing source code and rebuilding)?

```
 835 : Check Function level reset
       START
  (not)     Skipping for BDF - 0x100  Classcode is : 0x2
  (not)     Skipping for BDF - 0x10000  Classcode is : 0x2
       BDF 0x10000 is EP  config space addr 0xF0100000
  (not)     Skipping for BDF - 0x80000  Classcode is : 0x2
       BDF 0x80000 is EP  config space addr 0xF0800000
  (not)     Skipping for BDF - 0x90000  Classcode is : 0x1
       BDF 0x90000 is EP  config space addr 0xF0900000
       PCI_SM_02
                                       : Result:  PASS
       END
```

Sujana-M, 2024-10-21:
Hi @hrw,

The test 835 verifies the functionality of the device's function level reset (FLR). If the ACS initiates an FLR on an active Ethernet controller—handling data transmission and reception—the reset might disrupt the device's initialisation process, potentially leading to corruption.
Hence, skipped for Ethernet.

Thanks,
ACS Team

hrw, 2024-10-21:
If all network cards are simple ones (aka not smartnic running own OS) then how they do anything in EFI shell where there is no multitasking and the only running binary is bsa acs?

chetan-rathore, 2024-10-21:
Hi @hrw,

The changes in the test to skip network controller was made after feedback from partners who had ethernet port active and encountered issues while running of the test.
Since lot of partners might have the ACS integrated in the CI, we added the check to skip these devices to avoid any random crash.

We have taken your feedback on adding option to run the test on current skipped devices based on user input.

Thanks,
ACS team

closed 2024-10-21 by hrw