ARM-software/bsa-acs #368: unhandled synchronous exception   while running PE test

quic-pansing, 2024-10-24:
https://github.com/ARM-software/bsa-acs/commit/c266c563c71ce98f5380d6adfcb7785232bed53a  with above change we see panic while running PE test.  is chage are correct and validated before test

BSA Architecture Compliance Suite
          Version 1.0.8

 Starting tests with print level :  3

 Creating Platform Information Tables
 PE_INFO: Number of PE detected       :    8
 Primary PE: MIDR_EL1                 :    0x412FD050
  GIC CPUIF not present
 GIC INFO: GIC version                :    v3
 GIC_INFO: Number of GICD             :    1
 GIC_INFO: Number of GICR RD          :    1
 GIC_INFO: Number of ITS              :    1
 TIMER_INFO: System Counter frequency :    19 MHz
 TIMER_INFO: Number of system timers  :    7
 WATCHDOG_INFO: Number of Watchdogs   :    0
 PCIE_INFO: Number of ECAM regions    :    0
 SMMU_INFO: Number of SMMU CTRL       :    0
 Peripheral: Num of USB controllers   :    0
 Peripheral: Num of SATA controllers  :    0
 Peripheral: Num of UART controllers  :    0

      *** Starting PE tests ***

Operating System View:
   1 : Check Arch symmetry across PE         unhandled synchronous exception

chetan-rathore, 2024-10-24:
Hello @quic-pansing,

Yes the changes has been verified at our end on couple of systems.
Can you please share the exception logs to determine which register access has cause this issue.
Also are you run the ACS on top of baremetal system or edk2 based system.

Thanks,
ACS team

quic-pansing, 2024-10-25:
thanks @chetan-rathore 
    This is EDK2 based system. 
    I believe this mrs   x0, erridr_el1   this is creating "unhandled synchronous exception"

rajatgoyal47, 2024-11-08:
Hello Pankaj, 

Access to these registers is permitted only if FEAT_RAS is implemented. The ACS attempts to access the registers solely when FEAT_RAS is present. 

Additionally, access to Error Record registers from lower exception levels (such as EL2) requires Disabling the SCR_EL3.TERR bit to not trap the access. Please confirm that the relevant configuration is enabled.

Regards,
ACS Team

closed 2024-11-26 by quic-pansing