ARM-software/bsa-acs #379: TC 603 Check Arm BSA UART register offsets when using 16550 UART

f4bjh, 2024-11-05:
Hi,

When looking at [this documentation on page 21](https://github.com/ARM-software/bsa-acs/blob/v24.04_MEMTEST_REL1.0.0/docs/arm_bsa_architecture_compliance_test_scenario.pdf) , TC 603 seems to applicable with 16550 UART.

[image]

But when looking at [code](https://github.com/ARM-software/bsa-acs/blob/e4a78253cb06cfad33d1b7304e92ab4b390e27aa/test_pool/peripherals/operating_system/test_os_d003.c#L154), it does not seem to be the case

Note : I'm using tag v24.04_MEMTEST_REL1.0.0.

Is it possible to execute test TC 603 "Check Arm BSA UART register" if my hardware is using a 16550 UART ?

Thanks in advance for your help.

chetan-rathore, 2024-11-05:
Hi @f4bjh,

The rule id column is the test scenario document is aligned with the specification checklist, and if multiple tests are required for that rules, there will be multiple test id entries in Test number column.

The rule B_PER_05 is covered by test 603 and 606.
603 covers the generic uart and 606 covers the 16550 uart.

Thanks,
ACS team

f4bjh, 2024-11-06:
Hi,

Thank you very much for your answer, wich makes sense.

Can be closed.

Best regards

closed 2024-11-06 by f4bjh