ARM-software/bsa-acs #395: PC-BSA support?

hrw, 2024-11-18:
Arm released PC-BSA specification. System to be compliant needs to pass BSA ACS and some tests from SBSA ACS [1].

1. https://gpages.juszkiewicz.com.pl/sbsa-ref-status/bsa-sbsa.html

Are there plans to integrate PC-BSA testing into BSA ACS?

hrw, 2024-11-19:
BSA ACS, PC-BSA ACS, SBSA ACS...

Again no one thought that it may be a nice idea to have one ACS app rather than keeping same code in 3 places ;(

chetan-rathore, 2024-11-20:
Hi @hrw, 

You might have misread the comment above. PC-BSA ACS is name of the ACS not name of repository.

Thanks,
Chetan

hrw, 2024-12-17:
> ACS is planning a initial release of PC-BSA ACS in first week of December.

This year?

chetan-rathore, 2025-01-17:
Hi @hrw,

The PC-BSA ACS upstreaming to github is currently not planned, we will update you once its finalized.

Thanks