ARM-software/bsa-acs #405: Release v24.11_REL1.0.9   indentation error

quic-pansing, 2024-11-28:
line 352 : indentation error:
https://github.com/ARM-software/arm-systemready/blob/main/common/log_parser/apply_waivers.py

Log:

[   55.338703] init.sh[375]:   Waiver File        : /mnt/acs_tests/config/acs_waiver.json
[   55.697604] init.sh[381]:   File "/usr/bin/log_parser/apply_waivers.py", line 353
[   55.697811] init.sh[381]:     output_json_data = None

ricardosalveti, 2024-12-03:
Please propose a patch with the fix.

chetan-rathore, 2024-12-04:
Hi @ricardosalveti and @quic-pansing,

We will be upstreaming the fix for the issue by eod.

Thanks,
Chetan

chetan-rathore, 2024-12-04:
The changes for the fix is merged in arm-systemready repo: https://github.com/ARM-software/arm-systemready/pull/235

closed 2024-12-23 by chetan-rathore