ARM-software/bsa-acs #406: Release v24.11_REL1.0.9  Capsule update has multiple issues

quic-pansing, 2024-11-28:
https://github.com/ARM-software/arm-systemready/blob/main/common/uefi_scripts/startup.nsh

1.line 127: capsule_update_check.flag  is not part of test suite

2.line 162 : if exist FS%r:\acs_tests\app\unauth.bin then -> always validate unatuh.bin in every  boot  and need to bypass once it validated unatuh.bin  through some flag or file.

3.line:169: if exist FS%r:\acs_tests\app\tampered.bin then -> need similar logic to bypass once it validate tamperd.bin

4. line157: rm FS%r:\acs_tests\app\capsule_update_check.flag -> this condition is added before completion of capsule update .

5. line158:  echo "" > FS%r:\acs_tests\app\capsule_update_done.fla g-> this condition is added before completion of capsule update .

sathishas89, 2024-12-05:
Hi,
Please find my below comments:
1.line 127: capsule_update_check.flag is not part of test suite
capsule_update_check.flag is getting created as part of capsule update state machine from acs-linux.
For more details, please read Capsule update document from below link:
  https://github.com/sathishas89/arm-systemready/blob/main/common/docs/Automatic_Capsule_Update_guide.md
  
2.line 162 : if exist FS%r:\acs_tests\app\unauth.bin then -> always validate unatuh.bin in every boot and need to bypass once it validated unatuh.bin through some flag or file.
Yes, unatuh.bin update in evry boot can be optimized by adding user input. If user press any key, then we can skip unauth.bin update or else proceed to update.

3.line:169: if exist FS%r:\acs_tests\app\tampered.bin then -> need similar logic to bypass once it validate tamperd.bin
Yes, tampered.bin update in evry boot can be optimized by adding user input. If user press any key, then we can skip unauth.bin update or else proceed to update.

4. line157: rm FS%r:\acs_tests\app\capsule_update_check.flag -> this condition is added before completion of capsule update .
5. line158: echo "" > FS%r:\acs_tests\app\capsule_update_done.fla g-> this condition is added before completion of capsule update .
- Once Capsule update starts, CapsuleApp.efi checks header validation, if it is a valid capsule,
  then it copies same capsule to ESP partition and then it executes reset as part of capsule update.
- Since we dont have control on reset sequence of capsule update, I have created done flag and removed check flag just before reset.
- On next reboot, uboot firmware performs capsule update.

chetan-rathore, 2024-12-23:
Hi @quic-pansing,

Hope your queries are answered, we are closing the ticket. Please reopen if any further queries.

closed 2024-12-23 by chetan-rathore