ARM-software/bsa-acs #407: Release v24.11_REL1.0.9  Linux Boot up failure with initrd from gurb

quic-pansing, 2024-11-28:
device failed to boot due to prebuild image-> grub cfg -> initrd

               initrd /core-image-initramfs-boot-ub.cpio.gz

Log:    

5.410658] CPU4: Booted secondary processor 0x0000000400 [0x411fd411]
[    5.410775] Unable to handle kernel NULL pointer dereference at virtual address 0000000000000008
[    5.437913] Mem abort info:

chetan-rathore, 2024-11-28:
Hi @quic-pansing,

Can you share details which prebuild image you have used and any change made to grub menu at your side.

Thanks,
Chetan

smangali-qc, 2024-11-28:
HI chetan ,
prebuild image picked from

 The pre-built Release Image can be found at [https://github.com/ARM-software/arm-systemready/releases/download/v24.11_SR_REL3.0.0-BETA0_SR-DT_REL3.0.0-BETA0/systemready_acs_live_image.img.xz.zip](https://github.com/ARM-software/arm-systemready/releases/download/v24.11_SR_REL3.0.0-BETA0_SR-DT_REL3.0.0-BETA0/systemready-dt_acs_live_image.wic.xz.zip)
 
 
 menuentry 'Linux Boot'{
linux /Image root=PARTUUID=0235778b-b12e-45cb-8e03-462aee8d0c1b rootwait rootfstype=ext4 video=efifb:off
initrd /core-image-initramfs-boot-genericarm64.cpio.gz
}

thanks,
shiva

ricardosalveti, 2024-12-03:
This could be related to the firmware used, since this same image works fine with other devices.

chetan-rathore, 2024-12-04:
Hi @quic-pansing  and @smangali-qc,

Is there any particular reason to suspect the initrd grub change for the boot failure ?

Can you try the boot with previous prebuilt image on same system and firmware (https://github.com/ARM-software/arm-systemready/tree/main/IR/prebuilt_images/v24.03_2.1.1) and let us know your observations.

Thanks,
Chetan

chetan-rathore, 2024-12-23:
Hi @quic-pansing,

Any updates on running with previous image ?

Thanks,
Chetan

quic-pansing, 2025-01-30:
@chetan-rathore look like same issue https://github.com/ARM-software/arm-systemready/issues/252

chetan-rathore, 2025-03-24:
Hi @quic-pansing,

The #252 issue is crash during bsa linux test execution, the linux has booted and then during the execution of bsa test tool crash coming.

But the initial prints in this issue shared, the linux booting itself not happening due to initrd. 

In case if you are observing bsa linux issue and not initrd related,  the analysis and changes can be followed on #252 

Thanks,
Chetan

chetan-rathore, 2025-04-25:
Hi @quic-pansing, 

As the discussin related to Linux boot failure and the DMA test is happening on #252, closing this ticket as duplicate.