ARM-software/bsa-acs #412: apply_waivers.py   script issue

quic-pansing, 2024-12-06:
1:-  argparse is not defined

[   73.246436] init.sh[418]: Traceback (most recent call last):
[   73.246711] init.sh[418]:   File "/usr/bin/log_parser/apply_waivers.py", line 512, in <module>
[   73.246902] init.sh[418]:     main()
[   73.247032] init.sh[418]:   File "/usr/bin/log_parser/apply_waivers.py", line 496, in main
[   73.247192] init.sh[418]:     parser = argparse.ArgumentParser(description='Apply waivers to test suite JSON results.')
[   73.247314] init.sh[418]:              ^^^^^^^^
[   73.247461] init.sh[418]: NameError: name 'argparse' is not defined. Did you forget to import 'argparse'?yes

2:- apply_waivers is not defined

[   43.911179] init.sh[391]: Traceback (most recent call last):
[   43.911374] init.sh[391]:   File "/usr/bin/log_parser/apply_waivers.py", line 513, in <module>
[   43.911543] init.sh[391]:     main()
[   43.911668] init.sh[391]:   File "/usr/bin/log_parser/apply_waivers.py", line 510, in main
[   43.911862] init.sh[391]:     apply_waivers(args.suite_name, args.json_file, args.waiver_file, args.output_json_file)
[   43.911987] init.sh[391]:     ^^^^^^^^^^^^^
[   43.912152] init.sh[391]: NameError: name 'apply_waivers' is not definedyes

3:-   issue while applying waiver for MVP test cases

[   66.527525] init.sh[408]: Log parsed successfully. JSON output saved to /mnt/acs_results/acs_summary/acs_jsons/dt_kselftest.json
[   66.722870] init.sh[409]: ERROR: Waiver for TestSuite 'DTValidation' is missing 'Reason'. Skipping TestSuite-level waiver.
[   66.723110] init.sh[409]: Traceback (most recent call last):
[   66.723242] init.sh[409]:   File "/usr/bin/log_parser/apply_waivers.py", line 513, in <module>
[   66.723405] init.sh[409]:     main()
[   66.723525] init.sh[409]:   File "/usr/bin/log_parser/apply_waivers.py", line 510, in main
[   66.723826] init.sh[409]:     apply_waivers(args.suite_name, args.json_file, args.waiver_file, args.output_json_file)
[   66.724041] init.sh[409]:   File "/usr/bin/log_parser/apply_waivers.py", line 359, in apply_waivers
[   66.724828] init.sh[409]:     suite_level_waivers, testsuite_level_waivers, subsuite_level_waivers, testcase_level_waivers, subtest_level_waivers = load_waivers(waiver_data, suite_name)
[   66.724987] init.sh[409]:                                                                                                                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[   66.725130] init.sh[409]:   File "/usr/bin/log_parser/apply_waivers.py", line 95, in load_waivers
[   66.725378] init.sh[409]:     subtests = test_suite.get('TestCase', {}).get('SubTests', []) or test_suite.get('SubSuite', {}).get('TestCase', {}).get('SubTests', [])
[   66.725581] init.sh[409]:                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[   66.725722] init.sh[409]: AttributeError: 'list' object has no attribute 'get'

chetan-rathore, 2025-01-17:
Hi @quic-pansing,

The argparse is already fixed in mainline, the latest devicetree image will have the fix.
Kindly follow the build steps https://github.com/ARM-software/arm-systemready/tree/main/SystemReady-devicetree-band#steps-to-manually-build-image  for obtaining latest image.

The waiver guide (https://github.com/ARM-software/arm-systemready/blob/main/common/docs/waiver_guide.md) provides steps to how to apply waivers. There is sample waiver file also present https://github.com/ARM-software/arm-systemready/blob/main/common/docs/example_waiver.json

Thanks,

chetan-rathore, 2025-02-11:
Hi @quic-pansing,

Please reopen the ticket if any further information required.

Thanks,
Chetan

closed 2025-02-11 by chetan-rathore