ARM-software/bsa-acs #413: TC 903 : ATS Functionality Check

f4bjh, 2024-12-10:
Hi,

For TC 903 of SBSA test suite, `bound`variable is used in ALIGN_MEMORY : https://github.com/ARM-software/bsa-acs/blob/main/val/sbsa/src/sbsa_acs_iovirt.c#L206

But this variable has antagonist declaration : 

https://github.com/ARM-software/bsa-acs/blob/de0064098150651bb186c01cda74aef1b0923121/val/sbsa/src/sbsa_acs_iovirt.c#L25

https://github.com/ARM-software/bsa-acs/blob/de0064098150651bb186c01cda74aef1b0923121/val/common/src/acs_iovirt.c#L30

https://github.com/ARM-software/bsa-acs/blob/de0064098150651bb186c01cda74aef1b0923121/val/bsa/src/bsa_acs_iovirt.c#L29

Shouldn't it be `uint64_t bound` everywhere ?

Note that I did not test TARGET_BM_BOOT neabled, but it should not have impact here.

Thanks in advance for your help,
Best regards
Fabrice DECROP LONGET

SrikarJosyula, 2024-12-23:
Hi @f4bjh,
Thank you for pointing out the issue. We have now isolated the memory access alignment only to Baremetal environment.
Can you please pick the binary with updated changes and try it out from your end?
Binary at - [Bsa_ACPI_target.efi](https://github.com/SrikarJosyula/bsa-acs/actions/runs/12463828312/artifacts/2355122482)

Thanks,
ACS Team.

SrikarJosyula, 2025-01-08:
Hi @f4bjh,
Please let us know if you’ve had a chance to test the binary or if there are any updates on this. 

Thanks,
ACS Team.

f4bjh, 2025-01-08:
Hi,

Thank you very luch. Your binary is working fine.

Best regards
Fabrice DECROP LONGET

PR #417 (2025-01-13): Align memory access only for Baremetal
 - Resolves #413
 - Remove mem_align dependency for UEFI
commit c1287117 (2025-01-13): Align memory only for Baremetal (#417)

- Resolves #413
 - Remove mem_align dependency for UEFI
closed 2025-01-13 by chetan-rathore