ARM-software/bsa-acs #414: SIE error

quic-pansing, 2024-12-12:
continue with https://github.com/ARM-software/bsa-acs/issues/400

issue still exists on latest ACS 3.0, but olderv version do not have this issue. @amrathesh

amrathesh, 2024-12-12:
Hi @quic-pansing, 

Thank you for testing with latest ACS image, would need some more information to debug the issue faced, 

> which ACS image you are using "Systemready band" or "SystemReady Devicetree band"? 
> Are you using prebuilt images or compiling it from latest source?

smangali-qc, 2024-12-16:
HI @amrathesh ,

The SIE test suite is has few test which is modifying/tampering the existing DB and then if there is reset invoked in subsequent test , due to mismatch DB, device failed to load the image .

**test case : Verify signed DB append update (pass)**

**reboot test case:   Runtime services test: Test MOR and MORLOCK** 

**log:**

DxeImageVerificationLib: Image is signed but signature is not allowed by DB and SHA256 hash of image is not found in DB/DBX.
The image doesn't pass verification: VenHw(D1531D41-3F80-4091-8D0A-541F59236D66)/HD(1,GPT,F679732A-B936-4CD2-A183-AD01CB526AF2,0x800,0x7D000)/\EFI\BOOT\BOOTAA64.EFI
ERROR: C40000002:V03051002 I0 5A50AA81-C3AE-4608-A0E3-41A2E69BAF94

**expectation:**
after this test, device should continue to boot to grub and run remaining test  and device should be able  generate the report .
 
Regards,
shiva kumar

amrathesh, 2024-12-16:
Hi @smangali-qc, 

Thank you for the detailed clarification.  The test cases which tamper/modify KEK/dB/dBX restores them to ACS keys shipped with the image (used for initial key provisioning) by calling SecureBootVariableCleanup function [1][2]. Given that all variable update SCT tests passed, we shouldn't be observing the above issue.

I acknowledge a bug in the bbr-acs codebase where SecureBootVariableCleanup was not called for individual checkpoints, and a direct return was invoked instead, which might lead to wrong db values. I have raised a patch for the same[3], review and merge is awaited. 

[1] [VariableUpdatesBBTest.c#L161](https://github.com/ARM-software/bbr-acs/blob/3b8fe441be45fe31087dfc9d1eeeb0dda0c03fec/bbsr/sct-tests/SecureBoot/BlackBoxTest/VariableUpdatesBBTest.c#L161)
[2] [SecureBootVariableCleanup function](https://github.com/ARM-software/bbr-acs/blob/3b8fe441be45fe31087dfc9d1eeeb0dda0c03fec/bbsr/sct-tests/SecureBoot/BlackBoxTest/SecureBootBBTestSupport.c#L272)
[3] [Invoke SecureBootVariableCleanup in case of error](https://github.com/ARM-software/bbr-acs/pull/88)

Regards,
Amrathesh

amrathesh, 2024-12-19:
Hi @smangali-qc, 

We have fixed the bug mentioned in previous reply, please build latest Systemready ACS and give a fresh run on SUT and share if it fixed the issue your are facing.

Thanks and Regards,
Amrathesh

amrathesh, 2025-01-08:
Hi @quic-pansing , @smangali-qc,

Please share any updates on this, after testing with latest SystemReady ACS build which includes above mentioned fixes.

Thanks and Regards,
Amrathesh

quic-pansing, 2025-01-08:
issue is still not fixed.. we are checking if we can fix, will propose change

chetan-rathore, 2025-02-11:
Hi @quic-pansing,

Any updates on the patch or PR for this issue.

Thanks,
Chetan

chetan-rathore, 2025-03-24:
Hi @quic-pansing,

Please reopen the issue, once the changes are ready for sharing.

Thanks,
Chetan

closed 2025-03-24 by chetan-rathore