ARM-software/bsa-acs #419: BSA Test 835 "Check Function level reset" needs a PAL override to allow for shorter delay time for pre-silicon

cdwilde, 2025-01-14:
In pre-silicon, waiting 100 ms is a really long time and isn't necessary:
          status = val_time_delay_ms(100 * ONE_MILLISECOND);

It would be awesome to be able to configure at the PAL layer so this is runnable on pre-silicon.
          status = val_time_delay_ms(<PAL DEFINED VARIABLE> * ONE_MILLISECOND);

It should of course default to the 100.

kiranronad, 2025-01-20:
Thanks for the query. 
It is understandable that in pre-silicon 100ms waiting time is more but It is necessary to wait 100ms as per PCIe spec after FLR is triggered/initiated. 

PAL file need to be updated by partner only. please check anything can be handled here. 
pal/baremetal/target/RDN2/common/src/pal_bm_misc.c. 

yes, these tests should be in both BSA and SBSA.
Please let us know your feedback.

cdwilde, 2025-01-21:
This isn't a PAL file. The problematic code is here: test_pool/pcie/operating_system/test_os_p035.c (which the partner shouldn't update). 
test_pool/pcie/operating_system/test_os_p035.c:                  status = val_time_delay_ms(100 * ONE_MILLISECOND);
test_pool/pcie/operating_system/test_os_p035.c:                  status = val_time_delay_ms(ONE_MILLISECOND);

Where ONE_MILLISECOND is defined here:
val/common/include/pal_interface.h:#define ONE_MILLISECOND 1000

Since a partner shouldn't update these files, we need a way to make it feasible for pre-silicon tests. Or should we not run these tests in pre-silicon? 100 ms is even a long time on an emulator!

kiranronad, 2025-01-22:
Hi,

yes. Partner need not update "test_pool/pcie/operating_system/test_os_p035.c".
"val_time_delay_ms" function intern trigger pal interface "pal_time_delay_ms" in which partner need to update.
pal/baremetal/target/RDN2/common/src/pal_bm_misc.c

Ex:  Partner can implement multiplication factor to compensate delay
pal_time_delay_ms(uint64_t MicroSeconds)
{
  MicroSeconds = MicroSeconds * 0.1 ; //0.1- multiplication factor.
  // now "MicroSeconds" can be used for further implementation.
}

it reduces the delay of all the tests uses "val_time_delay_ms" with the factor.

we need to run these tests on pre-silicon. We have this test in BSA for Root port and End point devices and in SBSA for integrated devices(RCiEP and IEP_EP/ iEP_RP) 

Hope this solves your issue. Please let me know your feedback.

Thanks,
ACS Team

cdwilde, 2025-01-23:
This solution works! Being able to update it with pal_time_delay_ms is exactly what I needed. Thanks!

closed 2025-01-23 by cdwilde