ARM-software/bsa-acs #421: BSA Test 604 (Check Arm GENERIC UART Interrupt) needs a larger timeout

cdwilde, 2025-01-17:
The timeout in BSA Test 604 is too small (test_os_d003.c):

  val_set_status(index, RESULT_SKIP(TEST_NUM1, 2));
  while (count != 0) {
      **timeout = TIMEOUT_LARGE;**

Please move it to TIMEOUT_LARGE.

kiranronad, 2025-01-22:
Hi,

Thanks for the query. Partner can change the timeout value of TIMEOUT_MEDIUM in pal folder.
Please check by changing in below file and verify. 

Ex: pal/baremetal/target/RDN2/include/platform_override_fvp.h

#define TIMEOUT_LARGE    PLATFORM_BM_OVERRIDE_TIMEOUT_LARGE
#define TIMEOUT_MEDIUM   PLATFORM_BM_OVERRIDE_TIMEOUT_MEDIUM
#define TIMEOUT_SMALL    PLATFORM_BM_OVERRIDE_TIMEOUT_SMALL

/* Define the Timeout values to be used */
#define PLATFORM_BM_OVERRIDE_TIMEOUT_LARGE         0x10000
#define PLATFORM_BM_OVERRIDE_TIMEOUT_MEDIUM        0x1000
#define PLATFORM_BM_OVERRIDE_TIMEOUT_SMALL         0x10

Let me know if you need more information.

Thanks,
ACS Team

cdwilde, 2025-01-22:
Right, but this is more of a problem with the relative timeout compared to LARGE, MEDIUM, and SMALL. Lowering or raising these values affects all tests. In this particular test, LARGE makes more sense than MEDIUM (compared to all the other tests).

kiranronad, 2025-01-24:
Hi, 

I understand your point. 
what exactly issue observed here ? 
is SW not able to serve the interrupt within this time(MEDIUM) and test result return as fail ?

Thanks,
ACS Team

cdwilde, 2025-01-24:
Correct - 
**With TIMEOUT_MEDIUM**:
[11631666551 ps]  604 : Check Arm GENERIC UART Interrupt          
[11635449540 ps] ESR installing    
[11636130129 ps] ESR install result = 0x0    
[11639282527 ps]  pal_mmio_read A=0x30000004 D=0x17f041e    
[11649010352 ps] pal_gic_install_isr starting for int_id=112    
[11650571662 ps] pal_gic_install_isr core_num = 0    
[11651962756 ps] pal_gic_install_isr call to hw_int_registerHandler returned 0x0    
[11654534840 ps] pal_gic_install_isr complete, result is 0    
[11656216091 ps]  pal_mmio_write A=0x3000010c D=0x10000    
[11658227804 ps]     
[11658457990 ps]  pal_mmio_write16 A=0x2a400000 D=0xa    
[11774730818 ps]     
**[11774760873 ps]        Did not receive UART interrupt on 112**      
[11776452235 ps]        Failed on PE -    0  

**With TIMEOUT_LARGE**:
[8849092274 ps]  604 : Check Arm GENERIC UART Interrupt          
[8852895207 ps] Exception handler installing for ExceptionType 0    
[8854596680 ps] ESR install result = 0x0    
[8857739106 ps]  pal_mmio_read A=0x30000004 D=0x17f041e    
[8867466931 ps] pal_gic_install_isr starting for int_id=112    
[8869028103 ps] pal_gic_install_isr core_num = 0    
[8870419335 ps] pal_gic_install_isr call to hw_int_registerHandler returned 0x0    
[8873001391 ps] pal_gic_install_isr complete, result is 0    
[8874692753 ps]  pal_mmio_write A=0x3000010c D=0x10000    
[8876624274 ps]     
[8876854461 ps]  pal_mmio_write16 A=0x2a400000 D=0xa    
[9196700128 ps]     
**[9196730182 ps]        Received interrupt on 112** 

When we modify SMALL, MEDIUM, and LARGE for our tests to run in a pre-silicon environment, it shortens the times we need to complete tasks. Since these TIMEOUTs are all relative to each other, this particular test needs to be bucketed in the LARGE set.

PR #430 (2025-01-31): Fix for ARM GENERIC UART Interrupt Timeout - #421
BSA Test 604 (Check Arm GENERIC UART Interrupt) needs a larger timeout #421

Timeout is changed from **TIMEOUT_MEDIUM to TIMEOUT_LARGE.**

Fixes #421
closed 2025-01-31 by chetan-rathore