ARM-software/bsa-acs #487: 【BSA-Test】About linux bsa test-862“No extra address translation”

JackWeihuadi, 2025-05-07:
about DMA addr,as " _*dma_addr = bsa_dma_addr;  *dma_len  = bsa_dma_len;_"
We are using the address that has been applied for by the driver, but we cannot control when the driver is released.This led to the occasional failure of our 862 test.

_862: No extra address translation  
Device IOVA entry is fffff000 size = 1000 phys = 1803d0000  
Device IOVA entry is fffc0000 size = 20000 phys = 10009e6a4000  
Device IOVA entry is fffa0000 size = 20000 phys = 10009e6bc000  
Device IOVA entry is fff80000 size = 20000 phys = 10009e6d4000  
Device IOVA entry is fff60000 size = 20000 phys = 10009e6fc000  
Device IOVA entry is ffcdf000 size = 1000 phys = 1000affbc000  

The DMA address ffce0000 used by device  
is not present in the SMMU IOVA table  

Failed on PE - 0  
Checkpoint -- 0 : Result: FAIL_ 

How can this problem be avoided and is there a good solution

thanks

Sujana-M, 2025-05-22:
Hi @JackWeihuadi ,

Thanks for raising this query. Can you please share us the entire logs and the platform on which you are seeing this issue. 

Thanks,
ACS Team

JackWeihuadi, 2025-05-23:
Please refer to the attachment.

[Bsalog.txt](https://github.com/user-attachments/files/20405079/Bsalog.txt)

JackWeihuadi, 2025-05-29:
> Hi [@JackWeihuadi](https://github.com/JackWeihuadi) ,
> 
> Thanks for raising this query. Can you please share us the entire logs and the platform on which you are seeing this issue.
> 
> Thanks, ACS Team