ARM-software/bsa-acs #492: Handle VF's properly

ndhillonm, 2025-05-13:
Test case 802 for PCIe goes through ECAM but generates faux errors for virtual functions. I think this test should be modified to check if SR-IOV capability is present on a device and then all functions should be treated differently as per PCIe spec about virtual function config header.

Sujana-M, 2025-05-21:
Hi @ndhillonm,

Thank you for raising the issue. Can you please share the errors that you are getting and the list and type of the PCIe devices in your system.

Thanks,
ACS Team

ndhillonm, 2025-06-02:
802 : PE - ECAM Region accessibility check  
       START pal_mmio_read Address = 878040000000  Data = FFFFFFFF
 pal_mmio_read Address = 878040000100  Data = FFFFFFFF
 pal_mmio_read Address = 878040000FFC  Data = FFFFFFFF
 pal_mmio_read Address = 878040001000  Data = FFFFFFFF
....
 pal_mmio_read Address = 878000801000  Data = FFFFFFFF
 pal_mmio_read Address = 878000801100  Data = 1401000B

      Incorrect data for Bdf 0x80001    
       Failed on PE -    0
       PCI_IN_02
       Checkpoint -- 2048                           : Result:  FAIL
       END

In other words, test checks if given PCIE device (in this case Bdf 0x80001) has all 0xFFs within its PCIE config space when VendorID and DeviceID of that device returns 0xFFFFFFFF.
See actual test code: https://github.com/ARM-software/bsa-acs/blob/main/test_pool/pcie/operating_system/test_os_p002.c#L140
For Bdf 0x80001 (corresponding ECAM address is 0x878000801000 points to Vid/Did of device) VendorID and DeviceID is 0xFFFFFFFF.
However data at address 0x878000801100 is 0x1401000B (not 0xFFFFFFFF), thus test fails.

Shell> mm 878000801000 -w 4 -MMIO   
MMIO  0x0000878000801000 : 0xFFFFFFFF > 

MMIO  0x0000878000801004 : 0x00100004 > 

MMIO  0x0000878000801008 : 0x08800002 > 
...

MMIO  0x00008780008010FC : 0x00000000 > 

MMIO  0x0000878000801100 : 0x1401000B > 

MMIO  0x0000878000801104 : 0x040200A2 > 

...

MMIO  0x0000878000801200 : 0x00000000 > q

Shell>

chetan-rathore, 2025-07-16:
Closing this issue on bsa-acs as this is ported to new repository sysarch-acs: https://github.com/ARM-software/sysarch-acs